CabrilloRoboticsClub/cabrillo_rov_2023 #3: implement BME280 sensor in ROS

cTurtle98, 2022-10-19:
https://www.adafruit.com/product/2652
we will have two of this device.
one in the logic tube, and one in the thrust box.

implement a publisher in ros for the sensor data

sensor library docs:
https://learn.adafruit.com/adafruit-bme280-humidity-barometric-pressure-temperature-sensor-breakout/python-circuitpython-test

ros docs for writing publisher:
http://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Publisher-And-Subscriber.html

cTurtle98, 2022-10-19:
https://vscode.dev/github/CabrilloRoboticsClub/cabrillo_rov_2023/tree/3-implement-bme280-sensor-in-ros
open this branch is vs code web

DerVerrater, 2022-11-01:
ROS2 Python API docs.
https://docs.ros2.org/foxy/api/rclpy/api/node.html

cTurtle98, 2023-02-11:
working in branch i2c_proxy-to-ros-node

PR #23 (2023-02-11): I2c proxy to ros node
i2c_proxy is now a ros node.
closes #14 

bme280 is fully implemented
closes #3 

bno08x is disabled as it is not complete
closed 2023-02-11 by OrionOth