CabrilloRoboticsClub/cabrillo_rov_2023 #22: error linear_acceleration mus me type Vector3 on imu publisher

cTurtle98, 2023-02-11:
ERROR
```
ubuntu@SeaHawk:~/cabrillo_rov_2023$ ros2 launch ./launch/rov.yaml 
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-02-10-23-57-26-400634-SeaHawk-19981
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [h264_cam_node-1]: process started with pid [19982]
[INFO] [i2c_proxy-2]: process started with pid [19984]
[h264_cam_node-1] [INFO] [1676073447.458060022] [front_camera]: Parameter input_fn: /dev/video0
[h264_cam_node-1] [INFO] [1676073447.458894532] [front_camera]: Parameter fps: 20
[h264_cam_node-1] [INFO] [1676073447.458997697] [front_camera]: Parameter size: 800x600
[h264_cam_node-1] [INFO] [1676073447.459043974] [front_camera]: Parameter frame_id: rov_frame
[h264_cam_node-1] [INFO] [1676073447.459080604] [front_camera]: Parameter camera_info_path: info.ini
[h264_cam_node-1] [video4linux2,v4l2 @ 0xaaab11612180] Cannot open video device /dev/video0: No such file or directory
[h264_cam_node-1] [ERROR] [1676073447.459660802] [front_camera]: Could not open the v4l device: /dev/video0
[h264_cam_node-1] terminate called after throwing an instance of 'std::runtime_error'
[h264_cam_node-1]   what():  Could not open the v4l device
[ERROR] [h264_cam_node-1]: process has died [pid 19982, exit code -6, cmd '/home/ubuntu/cabrillo_rov_2023/install/h264_image_transport/lib/h264_image_transport/h264_cam_node --ros-args -r __node:=front_camera --params-file /tmp/launch_params_j8c6bvbe --params-file /tmp/launch_params_slqn7m32 --params-file /tmp/launch_params_ey3chm05 --params-file /tmp/launch_params_yt6bw64c -r image_raw/h264:=front_camera/h264'].
[i2c_proxy-2] Traceback (most recent call last):
[i2c_proxy-2]   File "/home/ubuntu/cabrillo_rov_2023/install/seahawk_rov/lib/seahawk_rov/i2c_proxy", line 33, in <module>
[i2c_proxy-2]     sys.exit(load_entry_point('seahawk-rov', 'console_scripts', 'i2c_proxy')())
[i2c_proxy-2]   File "/home/ubuntu/cabrillo_rov_2023/build/seahawk_rov/seahawk_rov/i2c_proxy/i2c_proxy.py", line 145, in main
[i2c_proxy-2]     rclpy.spin(node_i2c_proxy)
[i2c_proxy-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 222, in spin
[i2c_proxy-2]     executor.spin_once()
[i2c_proxy-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 713, in spin_once
[i2c_proxy-2]     raise handler.exception()
[i2c_proxy-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 239, in __call__
[i2c_proxy-2]     self._handler.send(None)
[i2c_proxy-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 418, in handler
[i2c_proxy-2]     await call_coroutine(entity, arg)
[i2c_proxy-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 332, in _execute_timer
[i2c_proxy-2]     await await_or_execute(tmr.callback)
[i2c_proxy-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 107, in await_or_execute
[i2c_proxy-2]     return callback(*args)
[i2c_proxy-2]   File "/home/ubuntu/cabrillo_rov_2023/build/seahawk_rov/seahawk_rov/i2c_proxy/i2c_proxy.py", line 127, in poll_sensors
[i2c_proxy-2]     message_logic_tube_imu.linear_acceleration = logic_tube_imu.raw_acceleration
[i2c_proxy-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/sensor_msgs/msg/_imu.py", line 297, in linear_acceleration
[i2c_proxy-2]     assert \
[i2c_proxy-2] AssertionError: The 'linear_acceleration' field must be a sub message of type 'Vector3'
[ERROR] [i2c_proxy-2]: process has died [pid 19984, exit code 1, cmd '/home/ubuntu/cabrillo_rov_2023/install/seahawk_rov/lib/seahawk_rov/i2c_proxy --ros-args -r __node:=i2c_proxy'].
ubuntu@SeaHawk:~/cabrillo_rov_2023$ 
```

i believe the important bit is this
```
[i2c_proxy-2] AssertionError: The 'linear_acceleration' field must be a sub message of type 'Vector3'
```

TODO:
find out what type the raw_acceleration is on the bno08x package.
transform that type to a vector3

cTurtle98, 2023-02-11:
this is holding #24

OrionOth, 2023-04-05:
Output of BNO085 is as follows:
```py
[seahawk_rov-1] Orientation  : (0.032958984375, 0.15380859375, -2.704833984375, 2.942626953125)
[seahawk_rov-1] Acceleration : (-0.54296875, -0.3203125, 7.984375)
[seahawk_rov-1] Rotation     : (0.001953125, 0.001953125, 0.0)
[seahawk_rov-1] Orientation  : (0.03076171875, 0.1552734375, -2.69970703125, 2.947265625)
[seahawk_rov-1] Acceleration : (-0.16796875, -0.125, 2.4453125)
[seahawk_rov-1] Rotation     : (0.0, 0.001953125, 0.001953125)
[seahawk_rov-1] Orientation  : (0.031005859375, 0.155029296875, -2.712890625, 2.93505859375)
[seahawk_rov-1] Acceleration : (-0.04296875, -0.046875, 0.53125)
[seahawk_rov-1] Rotation     : (0.0, 0.0, 0.001953125)
[seahawk_rov-1] Orientation  : (0.03173828125, 0.155029296875, -2.723876953125, 2.925048828125)
[seahawk_rov-1] Acceleration : (-0.01171875, 0.0078125, 0.1328125)
[seahawk_rov-1] Rotation     : (0.0, 0.0, 0.001953125)
[seahawk_rov-1] Orientation  : (0.031494140625, 0.155517578125, -2.723388671875, 2.925537109375)
[seahawk_rov-1] Acceleration : (-0.00390625, -0.0234375, 0.0)
[seahawk_rov-1] Rotation     : (-0.005859375, 0.0, -0.001953125)
[seahawk_rov-1] Orientation  : (0.03125, 0.155517578125, -2.721923828125, 2.9267578125)
[seahawk_rov-1] Acceleration : (0.03515625, -0.0234375, 0.02734375)
[seahawk_rov-1] Rotation     : (0.0, 0.0, 0.0)
[seahawk_rov-1] Orientation  : (0.031494140625, 0.15478515625, -2.721923828125, 2.9267578125)
[seahawk_rov-1] Acceleration : (-0.00390625, 0.015625, -0.0234375)
[seahawk_rov-1] Rotation     : (0.0, 0.0, 0.0)
[seahawk_rov-1] Orientation  : (0.031005859375, 0.154541015625, -2.72998046875, 2.91943359375)
[seahawk_rov-1] Acceleration : (-0.01171875, -0.0234375, 0.00390625)
[seahawk_rov-1] Rotation     : (0.0, 0.0, 0.0)
[seahawk_rov-1] Orientation  : (0.030517578125, 0.154541015625, -2.7197265625, 2.928955078125)
[seahawk_rov-1] Acceleration : (-0.01171875, 0.015625, 0.00390625)
[seahawk_rov-1] Rotation     : (0.0, 0.0, 0.0)
[seahawk_rov-1] Orientation  : (0.03173828125, 0.154052734375, -2.733154296875, 2.91650390625)
[seahawk_rov-1] Acceleration : (-0.01171875, 0.0078125, 0.00390625)
[seahawk_rov-1] Rotation     : (0.0, 0.0, 0.0)
[seahawk_rov-1] Orientation  : (0.03271484375, 0.154296875, -2.73974609375, 2.91015625)
[seahawk_rov-1] Acceleration : (-0.00390625, -0.03125, -0.03125)
[seahawk_rov-1] Rotation     : (0.0, 0.0, 0.0)
```

PR #88 (2023-04-13): [seahawk_rov] Fix BNO085 IMU message data type
The Adafruit BNO085 report datatypes being different from the ROS Imu message type has been an outstanding issue for some time now. Fortunately, with this destructuring, it is no longer an issue, and the IMU is now publishing messages that seem to be correct. Of course, the IMU working is different from assigning the correct variables to the correct places. We probably have to test this on the actual robot before we merge this, to publish in the correct order.

closes #22
closed 2023-04-13 by OrionOth