CabrilloRoboticsClub/cabrillo_rov_2023 #27: ROV Cameras

OrionOth, 2023-02-11:
- [ ] Publish camera views
- [ ] Subscribe to camera views
- [ ] Manipulate cameras

cTurtle98, 2023-02-11:
publishing of the rov is already implemented with the submodule mike added

OrionOth, 2023-03-10:
```py
[h264_cam_node-1] [video4linux2,v4l2 @ 0xaaab02718e10] Cannot find a proper format for codec 'h264' (id 27), pixel format 'none' (id -1)
[h264_cam_node-1] [ERROR] [1678314685.129901946] [front_camera]: Could not open the v4l device: /dev/video0
[h264_cam_node-1] terminate called after throwing an instance of 'std::runtime_error'
[h264_cam_node-1]   what():  Could not open the v4l device
```

OrionOth, 2023-03-17:
Status update: We need new cameras. The Arducam B0180 IMX219 we are using takes too much memory and only gives an uncompressed image stream that we can't handle properly on our 4gb Pi.

cTurtle98, 2023-04-15:
status update. we bought rpi cam v3 and they arent supported on ubuntu yet.

we bought rpi cam v2

closed 2023-04-21 by MegaMichaelXP