CabrilloRoboticsClub/cabrillo_rov_2023 #28: enable imx219 on ubuntu on raspberry pi with ansible

cTurtle98, 2023-02-11:
the imx219 camera module does not work out of the box

steps are needed to enable it

ubuntu 22.04 on pi 4

cTurtle98, 2023-02-11:
this needs to happen before #27 can happen

cTurtle98, 2023-02-11:
adding
```
dtoverlay=imx219
```
to `/boot/config.txt` is how you enable the camera we have on raspbian

but that file doesnt exist on ubuntu 22.04 on pi

OrionOth, 2023-03-08:
I did this on the test platform to `/boot/firmware/config.txt`, but it's still not working right. I'm now getting these errors:
```py
[h264_cam_node-1] [video4linux2,v4l2 @ 0xaaab02718e10] Cannot find a proper format for codec 'h264' (id 27), pixel format 'none' (id -1)
[h264_cam_node-1] [ERROR] [1678314685.129901946] [front_camera]: Could not open the v4l device: /dev/video0
[h264_cam_node-1] terminate called after throwing an instance of 'std::runtime_error'
[h264_cam_node-1]   what():  Could not open the v4l device
```

OrionOth, 2023-03-18:
Update: per #27 we are switching to use different cameras, so this may be changing.

cTurtle98, 2023-04-15:
pi cam v2 requires below in boot config to work
```
camera_auto_detect=0
display_auto_detect=1
start_x=1
```

@mike-matera whats the startx about

OrionOth, 2023-04-15:
I'm not sure what the `start_x` line means in the boot config file, but the command `startx` starts the X display server with a default config file. I think it likely the two are related somehow, though commonly X is started via the init system (`systemd`/`init`).

PR #110 (2023-04-23): Ansible fixes
* automated software setup
* automated software build
* fixed software depends for h264 image transport not being installed
* general ansible beautification

also updated setup docs

closes #111
closes #69 
closes #28
closed 2023-04-23 by DerVerrater