CabrilloRoboticsClub/cabrillo_rov_2023 #44: Split Kinematics code from Controller code

OrionOth, 2023-03-07:
This will be/was talked about in the March 10th 2023 meeting of the club, however doing this has a plethora of benefits. I don't believe that this counts as "falling victim to Java Syndrome" as one team member put it, especially considering we still don't seem to have consensus on future control scheme status.
Abstracting with the Twist even further than we already are is, I believe, a Good Idea™. 
This should probably be resolved before #31 and is a subset of #26

OrionOth, 2023-03-23:
This seems to be an ongoing issue. All we've done is moved where the programming is happening. I want to adopt a system more similar to what cturtle98 is working on in the `seahawk_rov` package, with everything happening in the `__main__.py` file with all the controller code imported by `__init__.py`.

cTurtle98, 2023-04-15:
@OrionOth any progress on this?

OrionOth, 2023-04-15:
No, this needs to be done but at the moment is not a super high priority, though it really should be. I'm focusing on #87 right now.

DerVerrater, 2023-05-07:
## The __main__.py file

The existence of a `__main__.py` file allows Python to treat the package as executable. e.g. `python -m seahawk_deck` instead of `python -m seahawk_deck.input_xbox_one`. In this case, there are 3 different modules that are each their own entry point. It doesn't make sense to have an entry point for the package this way.

It's possible to have multiple ROS.Node objects from a single executable. We could have a single entry point that spawns the nodes, but this doesn't get us any abilities we don't already have.

## The __init__.py file

This file is only required so that the Python runtime will recognize the containing folder as a package. It *can* contain code, but it doesn't have to. I haven't been able to figure out why exactly the `seahawk_rov` fails to execute without the imports in the `__init__.py` file. I *am* certain, however, that it's not because doing so is a mandatory action.

## Abstracting the Twist messages

I like the idea of passing a 6-tuple to the ROV. A `(x, y, z, rho, phi, theta)` that describes the desired linear and angular action which is then interpreted by the vehicle.

As far as "abstraction" goes, this is a higher level of abstraction -- it only expresses a desired change with no suggestions about what tools might be used to get there. It could be a 6 DoF thruster arrangement like ours, or a 2 DoF vehicle like a car (forward/backward, and yaw). We have the deck expose just one kind of spatial motion, and let the vehicle -- any vehicle -- figure out how it'll achieve that. The largest effect is that this would buy the Deck programming team the ability to be completely unconcerned with the ROV's thruster layout. However...

...This doesn't meaningfully improve our code. Sure the Deck programmers could forget about the thruster numbering, but that's not a huge deal. The buttons and what actions they do must still be more or less completely known on both ends. Further, even the side-effects of an implementation change are are negligible. Transmitting 8 floats for throttles is basically nothing; the 6-tuple is similarly trivial. The conversion math between the 6-tuple and the 8 throttles is a bit of arithmetic. There simply aren't any gains to be made.

To be clear, I support refactoring the code as I describe above. It's pretty easy, "more correct", and the benefits of doing so would be realized if we ever reused the code, had downstream users, or stacked more on top of this. But we won't reuse the code, we don't have any downstream, and there's not much to stack on the control scheme. This would be a largely unrealized improvement, and we should prioritize accordingly.

PR #62 (2023-03-15): Separate controller code & kinematics motor control code
This creates a new, separate node for handling controller input, splitting it apart from `motion_controller.py`.

closes #44
closed 2023-10-01 by mike-matera