CabrilloRoboticsClub/cabrillo_rov_2023 #48: Crash: i2c_proxy-class logic tube magnetometer channel index OOB

OrionOth, 2023-03-08:
Date: 2023-03-08 12:45 Wednesday March 8th 2023
Location: CIS Lab Development Testing Platform
Commit: `i2c_proxy-class/73befaa86419d8bb67c67cb114aff36896ce6ac6`
Replication steps:
1. `ros2 launch launch/rov.py`
2. Not reliably repeatable, however it happens often
```python
[i2c_proxy-2]   File "/home/ubuntu/cabrillo_rov_2023/install/seahawk_rov/lib/seahawk_rov/i2c_proxy", line 33, in <module>
[i2c_proxy-2]     sys.exit(load_entry_point('seahawk-rov', 'console_scripts', 'i2c_proxy')())
[i2c_proxy-2]   File "/home/ubuntu/cabrillo_rov_2023/build/seahawk_rov/seahawk_rov/i2c_proxy/i2c_proxy.py", line 214, in main
[i2c_proxy-2]     sensor_publisher = SensorPublisher(node_i2c_proxy, i2c)
[i2c_proxy-2]   File "/home/ubuntu/cabrillo_rov_2023/build/seahawk_rov/seahawk_rov/i2c_proxy/i2c_proxy.py", line 112, in __init__
[i2c_proxy-2]     self.logic_tube_bno085.enable_feature(adafruit_bno08x.BNO_REPORT_RAW_MAGNETOMETER)
[i2c_proxy-2]   File "/home/ubuntu/.local/lib/python3.10/site-packages/adafruit_bno08x/__init__.py", line 982, in enable_feature
[i2c_proxy-2]     self._process_available_packets(max_packets=10)
[i2c_proxy-2]   File "/home/ubuntu/.local/lib/python3.10/site-packages/adafruit_bno08x/__init__.py", line 787, in _process_available_packets
[i2c_proxy-2]     new_packet = self._read_packet()
[i2c_proxy-2]   File "/home/ubuntu/.local/lib/python3.10/site-packages/adafruit_bno08x/i2c.py", line 88, in _read_packet
[i2c_proxy-2]     self._update_sequence_number(new_packet)
[i2c_proxy-2]   File "/home/ubuntu/.local/lib/python3.10/site-packages/adafruit_bno08x/__init__.py", line 838, in _update_sequence_number
[i2c_proxy-2]     self._sequence_number[channel] = seq
[i2c_proxy-2] IndexError: list assignment index out of range
```

cTurtle98, 2023-03-08:
https://learn.adafruit.com/adafruit-9-dof-orientation-imu-fusion-breakout-bno085/python-circuitpython

> NOTE: The BNO85 seems to work best on the Raspberry Pi with an I2C clock frequency of 400kHz. 
> You can make that change by adding this line to your /boot/config.txt file:
> `dtparam=i2c_arm_baudrate=400000`

cTurtle98, 2023-03-08:
i2c instantiation might need to be adjusted as well
example code from the link above:
```python
import time
import board
import busio
import adafruit_bno08x
from adafruit_bno08x.i2c import BNO08X_I2C

i2c = busio.I2C(board.SCL, board.SDA, frequency=800000)
bno = BNO08X_I2C(i2c)
```

OrionOth, 2023-03-08:
> https://learn.adafruit.com/adafruit-9-dof-orientation-imu-fusion-breakout-bno085/python-circuitpython

From some research, the correct path for the `config.txt` file on Ubuntu is `/boot/firmware/config.txt`

cTurtle98, 2023-04-12:
does #88 close this?

OrionOth, 2023-04-14:
if we encounter this again, we can re-open this

closed 2023-04-14 by OrionOth