CabrilloRoboticsClub/cabrillo_rov_2023 #82: [BUG] BME280 crash

OrionOth, 2023-03-22:
**Describe the bug**
havent read the log yet, its a crash

**To Reproduce**
Steps to reproduce the behavior:
1. it kinda happened after running for a while i havent debugged idk
2. in stem center
3. crash

**Screenshots / Error messages**
```python
[seahawk_rov-1] Traceback (most recent call last):
[seahawk_rov-1]   File "/home/ubuntu/cabrillo_rov_2023/install/seahawk_rov/lib/seahawk_rov/seahawk_rov", line 33, in <module>
[seahawk_rov-1]     sys.exit(load_entry_point('seahawk-rov', 'console_scripts', 'seahawk_rov')())
[seahawk_rov-1]   File "/home/ubuntu/cabrillo_rov_2023/build/seahawk_rov/seahawk_rov/__main__.py", line 76, in main
[seahawk_rov-1]     rclpy.spin(node_seahawk_rov)
[seahawk_rov-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 222, in spin
[seahawk_rov-1]     executor.spin_once()
[seahawk_rov-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 713, in spin_once
[seahawk_rov-1]     raise handler.exception()
[seahawk_rov-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 239, in __call__
[seahawk_rov-1]     self._handler.send(None)
[seahawk_rov-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 418, in handler
[seahawk_rov-1]     await call_coroutine(entity, arg)
[seahawk_rov-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 332, in _execute_timer
[seahawk_rov-1]     await await_or_execute(tmr.callback)
[seahawk_rov-1]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 107, in await_or_execute
[seahawk_rov-1]     return callback(*args)
[seahawk_rov-1]   File "/home/ubuntu/cabrillo_rov_2023/build/seahawk_rov/seahawk_rov/__main__.py", line 70, in publisher
[seahawk_rov-1]     logic_tube_bme280.publish()
[seahawk_rov-1]   File "/home/ubuntu/cabrillo_rov_2023/build/seahawk_rov/seahawk_rov/logic_tube_bme280.py", line 60, in publish
[seahawk_rov-1]     msg_humidity.relative_humidity = self.bme.humidity
[seahawk_rov-1]   File "/home/ubuntu/.local/lib/python3.10/site-packages/adafruit_bme280/basic.py", line 252, in humidity
[seahawk_rov-1]     self._read_temperature()
[seahawk_rov-1]   File "/home/ubuntu/.local/lib/python3.10/site-packages/adafruit_bme280/basic.py", line 119, in _read_temperature
[seahawk_rov-1]     while self._get_status() & 0x08:
[seahawk_rov-1]   File "/home/ubuntu/.local/lib/python3.10/site-packages/adafruit_bme280/basic.py", line 152, in _get_status
[seahawk_rov-1]     return self._read_byte(_BME280_REGISTER_STATUS)
[seahawk_rov-1]   File "/home/ubuntu/.local/lib/python3.10/site-packages/adafruit_bme280/basic.py", line 303, in _read_byte
[seahawk_rov-1]     return self._read_register(register, 1)[0]
[seahawk_rov-1]   File "/home/ubuntu/.local/lib/python3.10/site-packages/adafruit_bme280/basic.py", line 314, in _read_register
[seahawk_rov-1]     return self._bus_implementation.read_register(register, length)
[seahawk_rov-1]   File "/home/ubuntu/.local/lib/python3.10/site-packages/adafruit_bme280/protocol.py", line 23, in read_register
[seahawk_rov-1]     i2c.write(bytes([register & 0xFF]))
[seahawk_rov-1]   File "/home/ubuntu/.local/lib/python3.10/site-packages/adafruit_bus_device/i2c_device.py", line 101, in write
[seahawk_rov-1]     self.i2c.writeto(self.device_address, buf, start=start, end=end)
[seahawk_rov-1]   File "/home/ubuntu/.local/lib/python3.10/site-packages/busio.py", line 174, in writeto
[seahawk_rov-1]     return self._i2c.writeto(address, memoryview(buffer)[start:end], stop=stop)
[seahawk_rov-1]   File "/home/ubuntu/.local/lib/python3.10/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 52, in writeto
[seahawk_rov-1]     self._i2c_bus.write_bytes(address, buffer[start:end])
[seahawk_rov-1]   File "/home/ubuntu/.local/lib/python3.10/site-packages/Adafruit_PureIO/smbus.py", line 314, in write_bytes
[seahawk_rov-1]     self._device.write(buf)
[seahawk_rov-1] OSError: [Errno 121] Remote I/O error
```

**Additional context**
running tests on claw branch

cTurtle98, 2023-04-12:
does #88 close this?

OrionOth, 2023-04-15:
No, this is an i2c error I think #86 is related

OrionOth, 2023-10-13:
edit: i think this is i2c clock out of sync thing

steph1111, 2024-05-10:
Note to myself: this is NOT the imu that's crashing. Try catch is on the wrong thing currently

steph1111, 2024-05-31:
```py
except OSError:
  self.node.get_logger().info("Warning: BME208 failed to publish (OSError)")
except:
  self.node.get_logger().info("Warning: BME208 failed to publish (other)")
```
From @mike-matera 
Maybe best to leave the generic `except`. But better would be to make sure the relaunch feature is set in the launch description.