CabrilloRoboticsClub/cabrillo_rov_2023 #86: Implement threading in seahawk_rov

steph1111, 2023-03-24:
See: 
https://realpython.com/intro-to-python-threading/#what-is-a-thread
https://docs.python.org/3/library/threading.html#thread-objects
The adafruit code is in C so threading will work

DerVerrater, 2023-03-27:
The Adafruit code (for CircuitPython and Blinka) is written in Python. The Circuitpython implementation itself is written in C, but we're not using it -- we're sitting on Blinka to provide the API, and the standard Python runtime. You can check their GitHub repos by searching for CircuitPython and sorting by popularity. Like [this](https://github.com/orgs/adafruit/repositories?q=CircuitPython&type=all&language=&sort=stargazers).

The C language doesn't have an inherent understanding of threads and instead relies on libraries (and often the OS).

What is the motivation of threading the module, and how should the work be split between the threads?

cTurtle98, 2023-03-27:
At the meeting last Friday we observed the thrusters to be extremely unresponsive and jittery.

If we disabled the sensors it went away.

We read through the source code for the bme280 library from adafruit and found that they are calling sleep while reading the temperature.

With python threading ony one thing can be using the python interpreter at a time but c code can run in the background

Since python sleep is written in c the idea was if we threaded the reading of the sensors the sleep while reading the temperature would now be non blocking

cTurtle98, 2023-03-27:
https://github.com/adafruit/Adafruit_CircuitPython_BME280/blob/main/adafruit_bme280/basic.py#L120

cTurtle98, 2023-04-15:
@steph1111 can we get some progress on this?

closed 2023-04-28 by OrionOth