CabrilloRoboticsClub/cabrillo_rov_2023 #99: systemd launch robot on boot

cTurtle98, 2023-04-15:
create systemd files to start robot code at boot

OrionOth, 2023-04-20:
https://roboticsbackend.com/make-ros-launch-start-on-boot-with-robot_upstart/
https://index.ros.org/p/robot_upstart/#humble

DerVerrater, 2023-04-25:
https://en.m.wikipedia.org/wiki/Upstart_(software)

Upstart has been in maintenance mode for almost 10 years. We're not getting a copy of Ubuntu that has it instead of Systemd, and we're not doing to dual-stack init systems -- especially a dead one.

MegaMichaelXP, 2023-04-28:
Added the service to seahawk-rov-spare.local. The manual launch seems to be working, but checking the service using systemctl status seahawk_rov_upstart.service gives the following messages:

Apr 28 22:36:58 SeaHawk-ROV-SPARE seahawk_rov_upstart-start[1110]: [seahawk_rov-2]   File "/home/ubuntu/.local/lib/python3.10/site-packages/busio.py", line 123, in init
Apr 28 22:36:58 SeaHawk-ROV-SPARE seahawk_rov_upstart-start[1110]: [seahawk_rov-2]     self._i2c = _I2C(portId, mode=_I2C.MASTER, baudrate=frequency)
Apr 28 22:36:58 SeaHawk-ROV-SPARE seahawk_rov_upstart-start[1110]: [seahawk_rov-2]   File "/home/ubuntu/.local/lib/python3.10/site-packages/adafruit_blinka/microcontroller/generic_linux/>
Apr 28 22:36:58 SeaHawk-ROV-SPARE seahawk_rov_upstart-start[1110]: [seahawk_rov-2]     self._i2c_bus = smbus.SMBus(bus_num)
Apr 28 22:36:58 SeaHawk-ROV-SPARE seahawk_rov_upstart-start[1110]: [seahawk_rov-2]   File "/home/ubuntu/.local/lib/python3.10/site-packages/Adafruit_PureIO/smbus.py", line 125, in __init>
Apr 28 22:36:58 SeaHawk-ROV-SPARE seahawk_rov_upstart-start[1110]: [seahawk_rov-2]     self.open(bus)
Apr 28 22:36:58 SeaHawk-ROV-SPARE seahawk_rov_upstart-start[1110]: [seahawk_rov-2]   File "/home/ubuntu/.local/lib/python3.10/site-packages/Adafruit_PureIO/smbus.py", line 151, in open
Apr 28 22:36:58 SeaHawk-ROV-SPARE seahawk_rov_upstart-start[1110]: [seahawk_rov-2]     self._device = open("/dev/i2c-{0}".format(bus), "r+b", buffering=0)
Apr 28 22:36:58 SeaHawk-ROV-SPARE seahawk_rov_upstart-start[1110]: [seahawk_rov-2] PermissionError: [Errno 13] Permission denied: '/dev/i2c-1'
Apr 28 22:36:58 SeaHawk-ROV-SPARE seahawk_rov_upstart-start[1110]: [ERROR] [seahawk_rov-2]: process has died [pid 1114, exit code 1, cmd '/home/ubuntu/cabrillo_rov_2023/install/seahawk_r>

closed 2023-10-01 by mike-matera