CabrilloRoboticsClub/cabrillo_rov_2023 #142: Properly handle shutdown in ROV threading code

OrionOth, 2023-05-24:
At the end of `seahawk_rov/__main__.py`'s main `try:` statement, there's this block of code:
```py
try:
    # Execute callbacks nodes as they become ready
    executor.spin()
finally:
    executor.shutdown()
    seahawk_rov.distroy_node()
```
It should be easy to tell what the problem is here: the `seahawk_rov` package does not contain a (misspelled) `distroy_node()` method. It seems like what this function is supposed to do would be to free up all used threads, but since they are handled through ROS I don't see why this is necessary. Do we get memory leaks? Do the threads keep running?
Finally, while testing Ciaran's `rov-cont-servo` continuous servo branch, I encountered this exception: 
```py
[seahawk_rov-2] Traceback (most recent call last):
[seahawk_rov-2]   File "/home/ubuntu/cabrillo_rov_2023/build/seahawk_rov/seahawk_rov/__main__.py", line 113, in main
[seahawk_rov-2]     executor.spin()
[seahawk_rov-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 279, in spin
[seahawk_rov-2]     self.spin_once()
[seahawk_rov-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 764, in spin_once
[seahawk_rov-2]     self._spin_once_impl(timeout_sec)
[seahawk_rov-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 761, in _spin_once_impl
[seahawk_rov-2]     future.result()
[seahawk_rov-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 94, in result
[seahawk_rov-2]     raise self.exception()
[seahawk_rov-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 239, in __call__
[seahawk_rov-2]     self._handler.send(None)
[seahawk_rov-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 418, in handler
[seahawk_rov-2]     await call_coroutine(entity, arg)
[seahawk_rov-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 332, in _execute_timer
[seahawk_rov-2]     await await_or_execute(tmr.callback)
[seahawk_rov-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 107, in await_or_execute
[seahawk_rov-2]     return callback(*args)
[seahawk_rov-2]   File "/home/ubuntu/cabrillo_rov_2023/build/seahawk_rov/seahawk_rov/bme280.py", line 70, in publish
[seahawk_rov-2]     msg_humidity.relative_humidity = self.bme.humidity
[seahawk_rov-2]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/sensor_msgs/msg/_relative_humidity.py", line 154, in relative_humidity
[seahawk_rov-2]     assert \
[seahawk_rov-2] AssertionError: The 'relative_humidity' field must be of type 'float'
[seahawk_rov-2] 
[seahawk_rov-2] During handling of the above exception, another exception occurred:
[seahawk_rov-2] 
[seahawk_rov-2] Traceback (most recent call last):
[seahawk_rov-2]   File "/home/ubuntu/cabrillo_rov_2023/install/seahawk_rov/lib/seahawk_rov/seahawk_rov", line 33, in <module>
[seahawk_rov-2]     sys.exit(load_entry_point('seahawk-rov', 'console_scripts', 'seahawk_rov')())
[seahawk_rov-2]   File "/home/ubuntu/cabrillo_rov_2023/build/seahawk_rov/seahawk_rov/__main__.py", line 116, in main
[seahawk_rov-2]     seahawk_rov.distroy_node()
[seahawk_rov-2] AttributeError: module 'seahawk_rov' has no attribute 'distroy_node'
[seahawk_rov-2] The following exception was never retrieved: cannot use Destroyable because destruction was requested
[ERROR] [seahawk_rov-2]: process has died [pid 1880, exit code 1, cmd '/home/ubuntu/cabrillo_rov_2023/install/seahawk_rov/lib/seahawk_rov/seahawk_rov --ros-args -r __node:=seahawk_rov'].
```
As you can see, while it starts with an (unrelated to this issue) error in the BME280 sensor code, it ends up snowballing into this major issue causing a complete crash of the node.

cTurtle98, 2023-05-28:
Currently if we crash while the pwm boards are outputting they continue to output.

We need to set all the outputs to `None` when we shutdown to disable them

DerVerrater, 2023-05-29:
Then there should be an exception handler that does so. The issue, as described, is a type error compounding into a syntax error. The first issue has been resolved with Orion's commit 94db9ab7, so it wont cause any more type-error exceptions. Any other exceptions are still uncaught and will execute the invalid code `seahawk_rov.destroy_node()`.

To fix *this* crash, we call the right function or call no function. I don't believe it's particularly important which, in this scenario.

If there's stuff that needs to occur to safely shutdown the vehicle -- such as halting the thrusters -- list them so they can be goals toward closing the issue.

closed 2023-06-02 by cTurtle98