CabrilloRoboticsClub/cabrillo_rov_2023 #147: [BUG] SAFETY motor E stop ideas

cTurtle98, 2023-06-02:
**Describe the bug**
we dont have a way to emergency stop the motors on the rov

**To Reproduce**
Steps to reproduce the behavior:
1. power up the pi and launch the rov node
2. pull the power to the pi only but not the thrust box while the thrusters are on
3. thrusters will continue to run

**Expected behavior**
I expect the motors to stop but that isnt how the i2c board works

**Screenshots / Error messages**
n/a

**Additional context**
it would be good to have a way to e-stop the thrusters

cTurtle98, 2023-06-02:
https://learn.adafruit.com/16-channel-pwm-servo-driver/pinouts#control-pins-2980405

> OE - Output enable. Can be used to quickly disable all outputs. When this pin is low all pins are enabled. When the pin is high the outputs are disabled. Pulled low by default so it's an optional pin!

the servo driver board has an output enable pin
if we find a way to pull that high it disables the motors

gelvinp, 2023-06-06:
Have the power for your motors run through a NO (normally open) relay, actuated by your pi. if it loses power, it'll shut off your motors.

cTurtle98, 2023-06-06:
The problem is more when the software crashes the pwm drivers continue to output whatever pwm freq they last recieved, but that might be fixed by my addition of __del__ functions

closed 2023-10-01 by mike-matera