CabrilloRoboticsClub/cabrillo_rov_2023 #191: Create throttle curves

steph1111, 2024-01-15:
Throttle curves are used in drone piloting to change the relationship between the position of the stick and the amount of throttle produced.  This can allow for increased control of the vehicle in certain situations.

Currently, the code has a linear ratio between the position of the sticks on the xbox to the amount of thrust the motors produce. For example if the stick is at 50 percent, 50 percent of the max throttle is produced by the motors
[image]

A throttle curve remaps the stick input to throttle values along a certain curve.  A throttle curve like the one below  remaps low stick input to even lower throttle values allowing for more control at reduced speeds
[image]

Write an implementation for throttle curves on the ROV. This should be done on the [pilot_input](https://github.com/CabrilloRoboticsClub/cabrillo_rov_2023/blob/remote_ros_params/src/seahawk/seahawk_deck/pilot_input.py) node by remapping each `joy_node` sick input to a curve before adding these values to the twist message. Something like the following
```py
twist_msg.linear.x  = thrt_crv(controller["linear_x"] )
```
In addition add functionality for choosing different throttle curves. These will be selected using keyboard presses, `0`, `1`, `2`, etc. These values should be taken from the `throttle_curve_choice` ROS parameter (which is remotely updated by the `keyboard` node. For now, do not worry about the exact curves to use, these will need to be tested in the water before we know what works best. Just start with the basic implementation

tinymassi, 2024-05-03:
This was finished and pushed to main yeehaw

steph1111, 2024-05-04:
I was working on the dashboard and found an bug with the throttle curve implementation. I created a branch, [thrt-crv-bug](https://github.com/CabrilloRoboticsClub/cabrillo_rov_2023/tree/thrt-crv-bug) with a `self.get_logger().info()` which should help you in finding the bug as well. This is another really weird bug because we looked at the ros graph utility and curves seemed to be happening?? Is there anything which you think could have changed @tinymassi

steph1111, 2024-05-04:
Update: The previous comment seems to have resolved itself? Maybe I was not running it properly. For now lets assume this is not a problem

closed 2024-05-04 by steph1111