CabrilloRoboticsClub/cabrillo_rov_2023 #225: Spinny thing™

steph1111, 2024-05-03:
## Task summary
Implement some code which spins the spinny thing™ bidirectionally given inputs from the left and right dpad.

steph1111, 2024-05-03:
The final iteration of the spinny thing™ differs significantly from the original plan. Initially, it was meant to be integrated into the I2C mega node, but we encountered difficulties in getting [ServoKit](https://docs.circuitpython.org/projects/servokit/en/latest/) to provide any recognizable output to any of the pins. As a result, we relocated it to an RPi GPIO pin, made it its own node, and utilized PWM to drive it. It spins.

tinymassi, 2024-05-03:
The spinny thing code has been written and merged to main yeehaw!

closed 2024-05-03 by tinymassi